twinearthsoftware/ArduinoSketchUploader #2: Support for Arduino Micro (ATMEGA32U4)

Jenik5, 2017-01-01:
Hello, would it be possible to add the support for Arduino Micro (ATMEGA32U4)? It would be very helpful.
Thanks in advance. Jenik5

christophediericx, 2017-01-01:
Hi Jenik5,

Thank you for your interest in this project. I just ordered a Micro off Amazon in order to be able to implement (and test) support for it (basically: AVR109 type bootloaders). Given delivery time (and finding some free time for implementation) you can likely expect this to be completed in approximately 2 to 3 weeks.

Jenik5, 2017-01-05:
Hi Christoph,

it is very kind from you. If you have some donation page, please let me know. I can cover at least your hardware expenses.

Jenik 5

christophediericx, 2017-01-11:
Hi Jenik5, 

Support for Arduino Micro (ATMega32U4 + the AVR109 protocol) has now been implemented. Let me know if you encounter any issues.

And no worries about hardware expenses, I can always use an extra Arduino!

closed 2017-01-11 by christophediericx

Jenik5, 2017-01-11:
Hi Christo.phe,

I tested it and works great.

Thank you

Jenik5