twinearthsoftware/ArduinoSketchUploader #3: Native uploader

Pimmetje, 2017-01-31:
Hi Chris,

Your project is the first i found that allows for native uploading of hex files. (Most are only wrappers around avrdude). Congrats for that.

As i play around with arduino a lot i needed a way to version my projects. But also to track what version was in a arduino after i uploaded it. 
I created a build server (with gitlab and gitlab-ci that will produce a hex as build files. The first lines printed on the serial port are it's project, and git hash so i can track the loaded project/version.
Second i implemented a python project that will use the API of gitlab to download the HEX files. Now i only need a good way to upload it in one click to my connected board. (i dislike the arvdude wrapper i use now, because i have to close the serial in order to use it).

Could u tell me how hard it was to implement the protocol to upload the hex file?
Or could i better implement the gitlab API in .Net ;)

Thanks for the project. I will play with it for sure.

christophediericx, 2017-02-01:
Hi Pimmetje,

Thank you for your interest in this library. As to the question how hard it is to implement the actual Arduino upload functionality: like most things in life, not very hard once you figure out exactly how.

There is a couple of logical things to do. First, you must parse the HEX file contents. I implemented this in [the following library](https://github.com/christophediericx/IntelHexFormatReader) (which gets used by the ArduinoSketchUploader). The HEX format really is quite straightforward, I found the Wikipedia article on it sufficient to understand its structure.

Then, depending on the Arduino used (or more specifically, the MCU and the specific bootloader ROM installed on said Arduino), you need to talk it's protocol. Typically, it is a rough (sometimes incomplete) version of an Atmel protocol. E.g. the Optiboot bootloader which is stock on the Arduino Uno R3 uses the STK500 (version 1) protocol. Once you find the documentation you can start implementing.

You might find [this blog post](http://www.diericx.net/post/upload-hex-file-from-csharp/) I wrote about implementing the protocol a useful read.

When you are playing around with the code, be aware that some of it is really quite sensitive, and that putting breakpoints in certain locations can cause the code to behave differently than when running in an uninterrupted fashion. 

Best regards,

Christophe

Pimmetje, 2017-02-01:
I should have asked the question differently ;). How long did it take to get the first version working for you?
Do you think it's useful to have a application that can download the hex file from a build server and upload it?

christophediericx, 2017-02-02:
The first version was probably up and running in an evening or 2-3 I think.

As far as the usefulness is concerned: that totally depends on the observer. I _personally_ have no need for such a program, but that might be completely different for you and/or other people. Grabbing the binary from the build server (1) and uploading it to an Arduino (2) sound like 2 distinct logical steps to script / configure in an automatic deployment solution perhaps (e.g. Octopus)? 

Such a solution might be overkill for your situation though, in which a simple script / program might be a good solution (depending on the exact use case).

Best regards,

Christophe

closed 2017-02-07 by christophediericx