twinearthsoftware/ArduinoSketchUploader #4: Upload second time fails

ChristofferBergkvist, 2017-03-05:
Following the code example to upload a sketch works perfectly when I first plug in the Arduino Uno into the USB port of my PC.

However if I try to upload the same or a different sketch one more time it fails. It does not even get caught in an exception but the program simply terminates.

Only workaround seems to be to unplug the USB cable and then it works again.

christophediericx, 2017-03-07:
Thank you for your interest in this library. I will try to find some time later this week (or at the latest next weekend) to see if I can reproduce and fix this issue. Do you use the Arduino IDE (or avrdude) in between uploads, or is the ArduinoSketchUploader the only program involved?

christophediericx, 2017-03-09:
Could you please provide a code sample that illustrates how you use the library exactly?

I have just tried to reproduce the issue (with the latest version - 2.3.0 and an UNO R3) but could not trigger it.

Configuring logging should show what is happening... If you add NLog (and NLog config) as nuget  packages to the project that uses the uploader library, you will automatically gain logging.

This is a sample NLog.config file that will direct the logs to be output to the console (you could also configure it to write to a file if preferred):

```
<?xml version="1.0" encoding="utf-8" ?>
<nlog xmlns="http://www.nlog-project.org/schemas/NLog.xsd"
      xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance">

  <targets>
    <target name="console" xsi:type="Console" />
  </targets>

  <rules>
    <logger name="*" minlevel="Debug" writeTo="console" />
  </rules>
</nlog>
```
If "Debug" is insufficient, you could up the logging level to "Trace". That way, you will be able to see every byte sent to and received from the Arduino.

closed 2017-05-25 by christophediericx