twinearthsoftware/ArduinoSketchUploader #5: Support for UWP (Universal Windows Platform)

ChristofferBergkvist, 2017-03-05:
Nuget package fails to install in UWP. What needs to be done to make it work in an UWP app?

christophediericx, 2017-03-07:
Thank you for your interest in this library. I will try to find some time later this week (or at the latest in the weekend) to investigate and fix this issue.

christophediericx, 2017-03-09:
Out of the box, UWP does not support linking of "standard" .NET framework dlls. For some dependencies I control (e.g. IntelHexFormatReader) I could probably add UWP as a target platform. But unfortunately, that's not possible for some crucial components elsewhere in the dependency chain:

- System.IO.Ports.SerialPort is only supported on the traditional .NET desktop frameworks
- SerialPortStream is only supported on the traditional .NET desktop frameworks

This means that - at least for the foreseeable future - linking to the ArduinoUploader library will require consuming it from a traditional .NET framework environment.

closed 2017-03-09 by christophediericx