twinearthsoftware/ArduinoSketchUploader #7: Unhandled Exception: System.IO.IOException: Device Error

fanis0990, 2017-05-17:
My device Arduino UnoR3 CH340G
```
C:\Users\cpp>\\VBOXSVR\VM_Share\ArduinoSketchUploader-2.4.0\ArduinoSketchUploade
r.exe -f \\VBOXSVR\VM_Share\keyFile -p COM3 -m UnoR3
2017-05-17 10:14:37.5776|INFO|ArduinoUploader.ArduinoSketchUploader|Starting Ard
uinoSketchUploader...
2017-05-17 10:14:37.6057|INFO|ArduinoUploader.ArduinoSketchUploader|Starting upl
oad process for file '\\VBOXSVR\VM_Share\keyFile'.
2017-05-17 10:14:37.6479|INFO|ArduinoUploader.BootloaderProgrammers.SerialPortBo
otloaderProgrammer|Opening serial port COM3 - baudrate 115200
2017-05-17 10:14:38.2948|INFO|ArduinoUploader.ArduinoSketchUploader|Establishing
 sync...

Unhandled Exception: System.IO.IOException: Device Error
   at RJCP.IO.Ports.SerialPortStream.WriteCheckDeviceError()
   at RJCP.IO.Ports.SerialPortStream.WriteCheck(Byte[] buffer, Int32 offset, Int
32 count)
   at RJCP.IO.Ports.SerialPortStream.Write(Byte[] buffer, Int32 offset, Int32 co
unt)
   at ArduinoUploader.BootloaderProgrammers.SerialPortBootloaderProgrammer.Send(
IRequest request)
   at ArduinoUploader.BootloaderProgrammers.OptibootBootloaderProgrammer.Establi
shSync()
   at ArduinoUploader.ArduinoSketchUploader.UploadSketch(IEnumerable`1 hexFileCo
ntents)
   at ArduinoUploader.ArduinoSketchUploader.UploadSketch()
   at ArduinoSketchUploader.Program.Main(String[] args)
2017-05-17 10:14:40.1791|INFO|ArduinoUploader.BootloaderProgrammers.SerialPortBo
otloaderProgrammer|Closing serial port...

```

gavalierm, 2017-05-21:
Is COM3 right port for you? I m not expert but this looks like wrong COM number.

fanis0990, 2017-05-23:
Of course, the correct one, everything works on the Mega16 controller.

christophediericx, 2017-05-25:
Thank you for your interest in this library.

Sorry it took me some time to react to this issue. 

It's a peculiar issue, which I can't replicate with my UNO (R3).

Judging from the stack trace, the exception is thrown from the SerialPortStream library (which is an one of the dependencies of the ArduinoSketchUploader). Judging [from the code](https://github.com/jcurl/SerialPortStream/blob/0ebe8e8856c5ae3920b916f2d0756c9aece3f73f/code/SerialPortStream.cs) the background serial communication thread has died for some reason when this exception gets thrown. Unfortunately, there is very little information exposed from the library for troubleshooting purposes.

I'll try to dig into the issue somewhat more next week.

christophediericx, 2017-06-14:
The issue at hand here definitely seems to be the `CH340` chip on the Arduino clone (since like you verified, the code works with the default `ATMega16U2` chip).

A couple of questions:

* Can you upload the same sketch using avrdude (or from the IDE)?
* Have you installed [a CH340 driver](http://sparks.gogo.co.nz/ch340.html) and is the device recognized correctly?

Just as a further read, a similar thread on the Arduino forum indicates you might also be experiencing [a corrupted bootloader or board reset issues](https://forum.arduino.cc/index.php?topic=381462.0).

fanis0990, 2017-06-14:
It was checked in 3 computers.
Also, with the chip the CH340 does not work.