twinearthsoftware/ArduinoSketchUploader #8: UWP Support

cmasseyTCI, 2017-05-31:
Do you have plans to make this run under Windows 10 IoT UWP Apps?

christophediericx, 2017-06-05:
Could you please add some more detail? Do you mean deploying an application to a Windows 10 IoT Core device?

cmassman, 2017-06-05:
Yes… I need to be able to upload a sketch file to the Arduino from a Raspberry Pi running windows 10 IoT UWP app.

christophediericx, 2017-06-06:
Support for that is not planned. I don't have the relevant hardware and I have had very little free time in the last couple of months, so this is not something I can commit to.

That being said, it should not be too hard to get it working based on the principles applied in this code base (the general protocol is straightforward). The main issue would be that this library currently depends on `System.IO.Ports.SerialPort` and `SerialPortStream`. Both have quirks, and are / were only available on a traditional Windows .NET Framework environment (some people were working on a .NET Core port of SerialPort, but it's definitely bleeding edge if it's already available).

You may be able to leverage some of the classes from the `Windows.Devices.SerialCommunication` namespaces in the UWP context.

closed 2017-06-14 by christophediericx

VladimirAkopyan, 2017-07-29:
I could help if you were to organise code in such a way that is obvious where I should be putting uwp related stuff

scout208, 2020-01-10:
I was able to make this work for UWP for my application. I'm using an Arduino Mega2560 so I only implemented it for that particular bootloader but it isn't too difficult. Just need to change the `RJCP.IO.Ports.SerialPort` object in `ArduinoBootLoaderProgrammer.cs` to type `Windows.Devices.Communication.SerialDevice` and update the `ReceiveNext` and `Send` methods to use the SerialDevice namespace methods instead and I can confirm that it all works as expected.

What are your guys particular Arduino models you are using? If you're using an Arduino Mega2560 you could just use my code, if something else then it would just need to be modified slightly for that bootloader probably but you could use my code as an example.

Let me know if it will help and I will push my stuff to a fork.