twinearthsoftware/ArduinoSketchUploader #9: Support for Atmega168

tomrosenback, 2017-06-19:
Is it possible to add support for the Nano with Atmega168 chip too?

christophediericx, 2017-06-20:
Thank you for your interest in this library.

One of the evenings later this week, I will verify if I can add support for this. It is likely that it is no problem to do so, although i might not have the corresponding hardware to test (once implemented, it should just work though - the changes required are mostly hardware configuration).

tomrosenback, 2017-06-21:
Would be great! Tested with NanoR3 and works perfect, would be great to have it working for the 168 aswell. Thanks!

christophediericx, 2017-06-22:
Support has been added (with the identifier `NanoR2`). 

Could you please try it out? I don't have the relevant hardware myself.

Thanks!

tomrosenback, 2017-06-22:
Tested and it works! Thank you!!

christophediericx, 2017-06-22:
Great, thanks for providing feedback!

closed 2017-06-22 by christophediericx