twinearthsoftware/ArduinoSketchUploader #11: Bad NuGet packaging for NLog dependency (v2.4.3)

JeremiePatron, 2017-06-26:
Hello,

The NuGet NLog dependency is wrong in release 2.4.3, N**l**og instead of N**L**og

Extract of ArduinoUploader.nuspec:
  `<dependency id="Nlog" version="4.0.0" />`

Can you repackaging the NuGet package , please ?

christophediericx, 2017-06-26:
Sure, I'll repackage, but I'm unsure what tangible effect this has (e.g. is nuget package matching not case-insensitive by default)?

christophediericx, 2017-06-26:
A new package ([ArduinoUploader 2.4.4](https://www.nuget.org/packages/ArduinoUploader/2.4.4) has been uploaded (it might take a couple of minutes before it gets indexed).

Thanks again for your interest in this library.

JeremiePatron, 2017-06-27:
It's good with the version 2.4.4 , thank you !

closed 2017-06-27 by JeremiePatron