twinearthsoftware/ArduinoSketchUploader #12: Environment.Exit on error

JeremiePatron, 2017-06-27:
Hello,

When i use ArduinoSketchUploader in my C# app , I don't want Environment.Exit on error but just a log or a Exception.
It's possible to add an option for this ?   

```
internal static void LogErrorAndQuit(string errorMessage) 
        {
            logger.Error(errorMessage);
            Environment.Exit(1);
        }
```

christophediericx, 2017-06-29:
Hi,

Sure, that's a valid use case when using as a library. I have a couple of very busy days ahead, but I will try to make the required changes over the weekend!

Thanks,

JeremiePatron, 2017-06-29:
Nice, I will wait your changes. 
Thank you.

christophediericx, 2017-07-13:
Hi @JeremiePatron,

Sorry for the wait! This has now been implemented: an exception in the library will log and throw an `ArduinoUploaderException` rather than causing an `Environment.Exit` (the consuming code should decide how to handle this exception). The version this behavior is published in is 2.4.5 (might take a 10 minutes or so for the package update to become available on nuget.org).

Thanks again for your interest in this library!

Christophe

closed 2017-07-13 by christophediericx

JeremiePatron, 2017-07-13:
Thank you.
It's working.