twinearthsoftware/ArduinoSketchUploader #13: Progress indicator

dollop80, 2017-08-22:
Hi,
Thank you for your ArduinoSketchUploader. It works perfect for my projects with ATmega1284p and ATmega328p. But is there a way to know the flashing progress?

christophediericx, 2017-08-24:
Hi,

Thank you for your interest in this project. I assume you are currently using the ArduinoUploader nuget package as a library? 

In that case, the easiest way to get "progress indication" (so to speak) is to hook up NLog (one of it's dependencies) in your consuming project. Depending on the configured level (`INFO`, `DEBUG`, `TRACE`), you will automagically get a lot of information being channeled into your configured log target (which can be anything NLog supports). 

An example of this can be seen in this repository in the command line ArduinoSketchUploader project (which consumes the ArduinoUploader library logging output, and redirects it to stdout - together with it's own output).

I hope this answers your question!

Christophe

dollop80, 2017-08-24:
Absolutely! Christophe, thank you for the hint.

closed 2017-08-24 by dollop80