twinearthsoftware/ArduinoSketchUploader #15: Replace NLog with LibLog or Microsoft.Extensions.Logging

andymac4182, 2017-09-16:
I was wondering if you would consider using a logging abstraction to allow better integration with applications

https://github.com/damianh/LibLog/wiki/Using
https://www.nuget.org/packages/Microsoft.Extensions.Logging

christophediericx, 2017-09-19:
Hi andymac4182,

Thank you for your interest in this library, and thank you for submitting a PR request too.

Could you please elaborate as to why choosing `LibLog` over `NLog` would allow you to achieve better application integration? Some of my initial arguments against switching would be:

* (In my honest opinion) it's relatively ugly to include an extra (big) code file in the project directly (which then needs to be maintained and updated in a non standard way).
* Using reflection and "assumptions" about the consuming logging framework (the way `LibLog` works) is (once again, in my honest opinion) kind of ugly too. A lot of "magic" at play.
* You still require a "regular" logging framework so the total dependency for this particular library / command line application still increases with 1 by using `LibLog`.
* Some applications already depend on `NLog` being used in this library. There really needs to be a compelling case / significant advantage in order to offset the risk of breaking external code (although I realize it should "just work".

andymac4182, 2017-09-19:
There was a couple of reasons for looking at `LibLog` over `NLog`.

* NLog doesn't support any version of netstandard at the moment in a released package just their prerelease packages
* Not all applications use NLog. LibLog allows people to use the logging framework they would like to use instead of being tied to your choice.
* It doesn't remove the ability to use NLog just increases the amount of people that can get good logs integrated into their logging platform.
* LibLog appears to be simpler to implement than `Microsoft.Extensions.Logging` and that would cause a much larger overhead I believe.
* The alternative would be to build an internal logging implementation with extension packages to each logging implementation similar to how MassTransit does it but that would be a large overhead to maintain.

christophediericx, 2017-09-21:
Thank you for providing more detail. I agree with your point that a library should not force a dependency on a logging framework (e.g. `NLog`) onto code consuming it.

Therefore, I have removed the `NLog` dependency and instead exposed a small `IArduinoUploaderLogger` interface which one can implement if one wants to subscribe to the log messages emitted by the library (fully optional).

I believe this allows for the best of both worlds, without introducing a dependency on `LibLog`.

andymac4182, 2017-09-21:
Thanks @christophediericx I will have a look once you publish the new nuget package.

andymac4182, 2017-09-21:
I just used your updated package. I noticed you fixed up the logger but didn't pull in the netstandard changes.

christophediericx, 2017-09-22:
That's correct. I'm definitely not opposed to targeting `NETStandard` soon, but at the moment no testing at all has been done to see if everything still works properly when e.g. consuming the library from a .NET Core application. There might be some hickups in `SerialPortStream` (since serial port communication code is typically far from "portable"). I'll try and see if I can test it out in one of the coming weekends, though.

andymac4182, 2017-09-22:
@christophediericx `SerialPortStream` targets Netstandard 1.5. Would you like me to add a example app that targets netcore2 as well?

christophediericx, 2017-09-22:
Sure, that would be helpful. Curious to see what quirks (if any) will show up. I'll be out a couple of days (for a national park camping trip) but I should be able to provide feedback on Sunday or at the latest Monday evening. Thanks.

andymac4182, 2017-09-22:
Closing this issue off since the logging is resolved 👍

closed 2017-09-22 by andymac4182