twinearthsoftware/ArduinoSketchUploader #21: Arduino Leonardo bricked

deivid77, 2017-11-22:
Hi, I've used your .NET library with a Leonardo board. 
This is the code I used with no errors:
```
var uploader = new ArduinoSketchUploader(
    new ArduinoSketchUploaderOptions()
    {
        FileName = @"C:\Users\david\Desktop\Blink.ino.hex",
        PortName = "COM11",
        ArduinoModel = ArduinoModel.Leonardo
    });

uploader.UploadSketch();

```
Now I get my board undetectable in my Device Manager.
I've tried in another computer, another port, another cable without success.

Do you have any clue?

Best regards

christophediericx, 2017-11-22:
Hi,

Thank you for your interest in this library.

The good new is, it's very unlikely that your board has been damaged by using the ArduinoSketchUploader code (which just uses the binary API to communicate with the built in bootloader).

That being said, your sketch contents itself may cause the board to be no longer "visible" (that is, apart from the couple of seconds at the start where the bootloader runs).

To start of the troubleshooting excercise, I would start of with following the suggestions in the following thread: https://arduino.stackexchange.com/questions/5178/arduino-leonardo-is-not-recognized-by-usb-cable-in-windows-8.

Best regards,

Christophe

deivid77, 2017-11-23:
Hi Cristophe,

Finally the board is alive again using another Arduino as ISP.
The thing is, how may I use your library in order to upload blink.ino sketch, for example?
(I wouldn't like to have an invisible Arduino again)

Do I have to upload when the bootloader runs (at the beginning). 

Best regards,

David

christophediericx, 2017-11-24:
Hi David,

I'm happy to hear your board is back! You can use the library to program your device any time. The library will reboot the device, which triggers the bootloader on the board for a couple of seconds. At that time communication using your bootloaders protocol will be initiated, and the sketch transferred.

The library should really not make the device invisible by any means, unless the code / sketch you are uploading (see link above) causes the device no longer to be enumerated (you probably want a `Serial.Begin` or similar in your code). In other words, you should get the exact same behavior as when programming from the IDE (which would use `avrdude` under the hood to program the board). 

Best regards,

Christophe

closed 2018-01-02 by christophediericx