twinearthsoftware/ArduinoSketchUploader #22: Difference encountered during verification, write failed

x4code, 2017-11-23:
hello, im getting a lot of this errors messages when uploading hex: 
 - "Difference encountered during verification, write failed"
Any ideas why? 

Thanks

christophediericx, 2017-11-24:
Hi,

Thank you for your interest in this library. After every page that gets written, there is a "verification" step that re-reads said page, and verifies the content lines up 100% with what was just written.

Could you please provide:

* Details on which Arduino you are using
* A full log (at the `TRACE` level)

Does programming from the IDE (or with `avrdude` directly) work for you?

Best regards,

Christophe

x4code, 2017-11-24:
Hello Chris, 
first i would like to say, that great work you have done with this project,, really love it.
i made an windows APP based on your "dll" but i still want to implement some threading with loading bar progress and logs viewer. (Soon)

Im using AtMega2560 and using arduino IDE works every time.
Using yours sometimes works, sometimes gives that error (a phew times)

btw, how can i get full logs? (read all logs messages i mean)

Thank you ,
keep the good work

x4code, 2017-11-27:
Using this bit of code for MEGA2560 on "ArduinoSketchUploader.cs" seems to work pretty good with no failures: 
```
case ArduinoModel.Mega2560:
                    {
                        mcu = new AtMega2560();
                        serialPortConfig = new SerialPortConfig(serialPortName, 115200,
                            new ResetThroughEnablingDtrBehavior(), new ResetThroughTogglingDtrRtsBehavior(250, 50, true), 250);
                        programmer = new Stk500V2BootloaderProgrammer(serialPortConfig, mcu);
                        break;
                    }
```

Cheers,
x4code

christophediericx, 2018-01-10:
Hi x4code,

Sorry for the delay in responding, I must have missed this update. It makes sense to increase the timeouts a little, and it would also nicely explain the write failure. I'll integrate those changes. Thanks for investigating!

Christophe

closed 2018-01-10 by christophediericx