twinearthsoftware/ArduinoSketchUploader #24: Verification exception when uploading to Mega2560 board.

mightoz, 2018-01-15:
I am getting the exception: "Difference encountered during verification, write failed." When attempting to upload to a board with mega 2560.  I am using .NET 4.5.2, and ArduinoUploader 3.1. 

There are no expections if i upload firmware with avrdude and powerterminal, and then use Arduino Uploader with the same hex file. But if I try to upload the .hex file that is generated by Arduino ide a still get them problem. 

Any clues?

christophediericx, 2018-01-19:
In order to diagnose this problem, we need to gather a bit more information:

* Could you please run a `TRACE` level log file (see the [logging section of the readme](https://github.com/christophediericx/ArduinoSketchUploader#logging)) , and post an extract of the section around the exception?
* What command line parameters are you passing to `avrdude` specifically for a working upload?

mightoz, 2018-01-23:
Alright, here is a trace level snippet

2018-01-23 10:05:42.9290 ArduinoSketchUploader Sending 10 bytes: 
1B-29-00-04-0E-14-01-00-20-0D
2018-01-23 10:05:42.9759 ArduinoSketchUploader Receiving byte: 1B
2018-01-23 10:05:42.9759 ArduinoSketchUploader Received MESSAGE_START.
2018-01-23 10:05:42.9759 ArduinoSketchUploader Receiving byte: 29
2018-01-23 10:05:42.9915 ArduinoSketchUploader Received sequence number: 41.
2018-01-23 10:05:43.0071 ArduinoSketchUploader Receiving byte: 01
2018-01-23 10:05:43.0071 ArduinoSketchUploader Receiving byte: 03
2018-01-23 10:05:43.0227 ArduinoSketchUploader Received message size: 259.
2018-01-23 10:05:43.0227 ArduinoSketchUploader Receiving byte: 0E
2018-01-23 10:05:43.0227 ArduinoSketchUploader Received TOKEN.
2018-01-23 10:05:43.0384 ArduinoSketchUploader Receiving bytes: 14-00-79-74-65-73-3A-20-00-20-46-72-65-65-20-4D-65-6D-6F-72-79-3A-20-00-40-61-62-20-20-30-20-32-30-31-30-00-43-6F-6D-70-69-6C-65-64-3A-20-00-28-6E-6F-6E-65-2C-20-64-65-66-61-75-6C-74-20-63-6F-6E-66-69-67-29-00-20-7C-20-41-75-74-68-6F-72-3A-20-00-40-61-62-20-20-30-20-32-30-31-30-20-31-32-3A-30-38-3A-33-34-00-20-4C-61-73-74-20-55-70-64-61-74-65-64-3A-20-00-31-2E-30-2E-32-00-4D-61-72-6C-69-6E-00-20-53-6F-66-74-77-61-72-65-20-52-65-73-65-74-00-20-57-61-74-63-68-64-6F-67-20-52-65-73-65-74-00-20-42-72-6F-77-6E-20-6F-75-74-20-52-65-73-65-74-00-20-45-78-74-65-72-6E-61-6C-20-52-65-73-65-74-00-50-6F-77-65-72-55-70-00-73-74-61-72-74-00-22-00-65-6E-71-75-65-69-6E-67-20-22-00-22-00-65-6E-71-75-65-69-6E-67-20-22-00-01-01-FF-00-00-A0-40-00-00-A0-40-00-00-00-40-00-80-10-43-00-00-00
2018-01-23 10:05:43.0384 ArduinoSketchUploader Receiving byte: 18
2018-01-23 10:05:43.0540 ArduinoSketchUploader Resetting...
2018-01-23 10:05:43.0540 ArduinoSketchUploader Toggling DTR/RTS...
2018-01-23 10:05:43.3975 ArduinoSketchUploader Closing serial port...
2018-01-23 10:05:43.3975 ArduinoSketchUploader Difference encountered during verification, write failed!

The command I passed to avrdude was

'.\avrdude -C.\avrdude.conf -v -patmega2560 -cwiring -PCOM3 -b115200 -D -Uflash:w:./Marlin.cpp.hex:i'

Hope this helps

christophediericx, 2018-02-14:
This should be fixed in 3.2 (by d94afa0331d5cca00a22caaa39d031d012852fed).

mightoz, 2018-02-25:
Indeed, it works!

closed 2018-02-25 by mightoz