twinearthsoftware/ArduinoSketchUploader #32: Add support for Arduino Nano R3 with new bootloader (optiboot)

SoftwareBySebastianLang, 2018-05-09:
The Arduino development team has replaced the bootloader for the Arduino NANO R3 boards in early 2018.

> NOTE: We have updated the NANO board with a fresh bootloader. Boards sold from us from January 2018 have this new bootloader, while boards manufactured before that date have the old bootloader. First, make sure you have the Arduino AVR Core 1.16.21 or later looking at the Board Manager. Then, to program the NEW Arduino NANO boards you need to chose Processor > "ATmega328P". To program old boards you need to choose Processor > "ATmega328P (Old Bootloader)". If you get an error while uploading or you are not sure which bootloader you have, try each type of processor 328P until your board gets properly programmed.
[Source](https://www.arduino.cc/en/Guide/ArduinoNano)

The previous bootloader ATmegaBOOT was exchanged for Optiboot and thus also doubled the data transfer rate from 57,600 Bd to 115,200 Bd. See comparison:

> **Arduino Nano w/ ATmega328P**
nano.menu.cpu.atmega328=ATmega328P
nano.menu.cpu.atmega328.upload.maximum_size=30720
nano.menu.cpu.atmega328.upload.maximum_data_size=2048
nano.menu.cpu.atmega328.upload.speed=**_115200_**
nano.menu.cpu.atmega328.bootloader.low_fuses=0xFF
nano.menu.cpu.atmega328.bootloader.high_fuses=0xDA
nano.menu.cpu.atmega328.bootloader.extended_fuses=0xFD
nano.menu.cpu.atmega328.bootloader.file=optiboot/optiboot_atmega328.hex
nano.menu.cpu.atmega328.build.mcu=atmega328p
**Arduino Nano w/ ATmega328P (old bootloader)**
nano.menu.cpu.atmega328old=ATmega328P (Old Bootloader)
nano.menu.cpu.atmega328old.upload.maximum_size=30720
nano.menu.cpu.atmega328old.upload.maximum_data_size=2048
nano.menu.cpu.atmega328old.upload.speed=**_57600_**
nano.menu.cpu.atmega328old.bootloader.low_fuses=0xFF
nano.menu.cpu.atmega328old.bootloader.high_fuses=0xDA
nano.menu.cpu.atmega328old.bootloader.extended_fuses=0xFD
nano.menu.cpu.atmega328old.bootloader.file=atmega/ATmegaBOOT_168_atmega328.hex
nano.menu.cpu.atmega328old.build.mcu=atmega328p

I think the easiest way would be to implement a new variation of [ATMega328P.cs](ArduinoSketchUploader/Source/ArduinoUploader/Hardware/ATMega328P.cs) and add the "new" Arduino model to the list in [ArduinoModel.cs](ArduinoSketchUploader/Source/ArduinoUploader/Hardware/ArduinoModel.cs) as _`NanoR3New`_ just like in the Arduino IDE:
[image]
[Image source](https://forum.digikey.com/t/january-2018-arduino-nano-bootloader-update/1194)

SoftwareBySebastianLang, 2018-06-28:
I just did some tests by modifying the library from the latest release [v.3.2.0](https://github.com/christophediericx/ArduinoSketchUploader/releases/tag/v3.2.0).

I got a successful result by adding the following code to [ArduinoUploader.xml](https://github.com/christophediericx/ArduinoSketchUploader/blob/master/Source/ArduinoUploader/ArduinoUploader.xml):
```
    ...
    <Arduino model="NanoR3New">
      <Mcu>AtMega328P</Mcu>
      <BaudRate>115200</BaudRate>
      <PreOpenResetBehavior>DTR;true</PreOpenResetBehavior>
      <CloseResetBehavior>DTR-RTS;250;50</CloseResetBehavior>
      <SleepAfterOpen>250</SleepAfterOpen>
      <ReadTimeout>1000</ReadTimeout>
      <WriteTimeout>1000</WriteTimeout>
      <Protocol>Stk500v1</Protocol>
    </Arduino>
    ...
```

Then I added the new model to the list in [ArduinoModel.cs](https://github.com/christophediericx/ArduinoSketchUploader/blob/master/Source/ArduinoUploader/Hardware/ArduinoModel.cs):
```
        ...
        NanoR3,
        NanoR3New,
        UnoR3
        ...
```

I have tested the code with five original Arduino NANOs with the Optiboot bootloader and got no errors. So @christophediericx could you please add this additions to your library with the next release?
Thank you.

Andre-Gloria, 2018-11-15:
Without taking merit from @SoftwareBySebastianLang since he points out, imho, the correct way of supporting the new chip change!
I would like to indicate a **_workaround_** that enables to use the current release [v.3.2.0](https://github.com/christophediericx/ArduinoSketchUploader/releases/tag/v3.2.0) with an Arduino Nano with Optiboot booloader until a new release emerges.
Since [ArduinoSketchUploader](https://github.com/christophediericx/ArduinoSketchUploader) only programs the device's flash one can use `--model=UnoR3` in order to successfully program a Nano.
This works because, as pointed out by @SoftwareBySebastianLang, the major change with the new bootloader is the programming baudrate.

Keep up the good work!
Kind regards.

PS: @SoftwareBySebastianLang, consider doing a pull request with your changes to better help @christophediericx incorporating them ;)

rickyelqasem, 2020-12-12:
Sebastian do you know how to add the support for an Atmega4808 (Thinary board) https://www.avrfreaks.net/forum/nano-every-use-atmega4808?page=all ? @SoftwareBySebastianLang