twinearthsoftware/ArduinoSketchUploader #49: Upload hex file using Visual Basic app

mily1994, 2019-12-27:
Hi
I'm trying to create VB application which will upload hex file using ArduinoSketchUploader NuGet. 
Here is my code:
Public Sub Sketch_Uploader(sender As Object, e As EventArgs)
        Dim folderPath As String
        folderPath = "C:\Users\Milosz\Desktop\"
        Dim fileName As String
        fileName = "blink.hex"
        Dim fullPath As String
        fullPath = My.Computer.FileSystem.CombinePath(folderPath, fileName)
        Dim uploader As ArduinoSketchUploader = Nothing
        Dim options As ArduinoSketchUploaderOptions = Nothing
        options.FileName = fullPath
        options.PortName = "COM3"
        uploader.UploadSketch()
End Sub
Code is being compliled but nothig happens. Do you have some idea what can be wrong with it or maybe it is not possible to do it using VB and i have to switch to C#.
 I would be very grateful for any help.

joshuag1000, 2020-02-01:
Are your options actually being applied? 
you have assigned the variable options your settings but i dont see where those settings are being given to the uploader variable.

joshuag1000, 2020-02-01:
Incase you are interested i am coding a gui for this uploader and can be found at https://github.com/SuperNinja-4965/Arduino-Hex-Uploader